pdx-robotics/ElectricSpeedController #1: Test Components

PrestonCazier, 2018-08-08:
- [ ] Test OLED Screen, using default libraries for functionality and general "it works"edness
- [ ] Test 4700uF capacitor to see if they are within tolerance
- [ ] Test 100uF capacitor to see if they are within tolerance
- [ ] Test smd capacitors to see if they are within tolerance
- [ ] Test resistors to see if they are within tolerance
- [ ] Test potentiometer to verify leads

PrestonCazier, 2018-08-14:
Amazon item page: https://www.amazon.com/gp/product/B0761LV1SD/ref=oh_aui_detailpage_o01_s01?ie=UTF8&psc=1

says that the OLED screen should work with Adafruit_GFX and Adafruit_SSD1306
to do the testing we will have to find the correct library and learn to use it

the goal of the OLED screen in the project is to have a display that will show the speed that the brushless motor is set to by the potentiometer 

useful links:
https://learn.adafruit.com/monochrome-oled-breakouts/arduino-library-and-examples
http://www.miniarduino.com/makerfocus-2pcs-i2c-oled-display-module-0-91-inch-i2c-ssd1306-oled-display-module-blue-i2c-oled-screen-driver-dc-3-3v5v-for-arduino/

PrestonCazier, 2018-08-14:
For the SMD components, a test case has been made but DO NOT TEST/OPEN/DO ANYTHING with these components yet

PrestonCazier, 2018-08-14:
The final test that can be done right now involves the Pot, we need to verify the leads, the outside two should be the through line and the middle one should be how we read the change in voltage/resistance.  this test also involves verifying that this is a 10k pot.  what is its maximum resistance

fill in test results [here](https://github.com/pdx-robotics/ElectricSpeedController/wiki/Component-Test:-Potentiometer)

PrestonCazier, 2018-08-14:
For the power jack we need more information on how to get it to work in a circuit to provide power.  what are the 3 leads for?  
[Amazon item page](https://www.amazon.com/gp/product/B00W944ACE/ref=oh_aui_detailpage_o01_s02?ie=UTF8&psc=1)  
Fill in test results [here](https://github.com/pdx-robotics/ElectricSpeedController/wiki/Component-Test:-Power-Jack)

PrestonCazier, 2018-08-14:
also please upload all datasheets and the correct libraries for the OLED to this repo

Chana030102, 2018-08-14:
100 and 100k ohm resistor bags measured and marked by @ianeyk

Chana030102, 2018-08-14:
Remeasured and multi-meter says 102K.

PrestonCazier, 2018-08-14:
thats good to know, also i want to congrads you two on being able to read my comments especially the one that said not to do the SMD

PrestonCazier, 2018-08-14:
also im thinking we maybe should just do through hole components

ianeyk, 2018-08-14:
For the power jacks: Lead 2 is always connected to the power prong (in the middle of the hole). Lead 1 is always connected to the ground tab (bottom of the hole/outside of power plug). When nothing is plugged in, leads 1 and 3 are connected; when the plug is in, the connection is broken.

ianeyk, 2018-08-14:
For the LED: when the leads are facing you and the LED head is facing up, the lead on the left is positive, and the one on the right is negative.